smagicld/xiaoxiangBMS #186: Switching on and off the battery poles via app does not work

JohnnyTig, 2022-05-19:
Today I received a new lifepo battery.

I recorded a video where you can see that the function of the buttons „battery on“ and „battery off“ does not work.

don’t ask me why the BMS version is not displayed correctly. I think it’s XiaoxiangBMS. And a lot of other functions are working (reading BMS data …).

smagicld, 2022-05-19:
This is because bms is protected by HW pin. Go into bms settings. Once asked for PIN - enter it. This will store the pin in the app so you do not have to enter it next time. The same pin is used then for switching the input/ output mosfet.

JohnnyTig, 2022-05-19:
Hi smagicld, thank you very much for your quick reply.

However, the app has never actively asked me for a password with this battery.

I have looked at the settings and cannot find any type of password request here either.

However, at one point it says that the firmware is too old for a hardware pin (see screenshot).

The only thing I have noticed is that when I write changes to the BMS, they are not saved. For example, if I deactivate the option "Hardware switch" (there is no hardware switch). After a few more readouts, the option is enabled again.

You can find the operating instructions for the battery here https://belluna.eu/bt/.

Am I missing something
[image]
[image]
?

JohnnyTig, 2022-05-20:
there is news, but unfortunately I still don't have a solution to my problem.

The supplier of the battery told me that the iOS app "Xiaxiang BMS" no longer works with the battery I received from him.

Instead, I should use his (?) Android app, which you can download here: https://belluna.eu/bt/ "bms.apk". 

However, I am not planning to switch from iOS to Android, so I am interested to know if there will be a solution for the Xiaoxiang app.

smagicld, 2022-05-20:
very good hint - probably they use customised XiaoXiang Boards / Firmware and blocked the function to switch off the battery.
Imagine: your battery config is accessible without any protection! everyone would be able to switch your battery off.

Questions:
1. which battery do you have? (link to the shop)
2. does only mosfet on/off not work with my app? (what about rest of the settings?)
3. have you tried the android app if you can switch on/off the battery? (if yes, then it is not blocked in Firmware and I could reverse engineer it)

JohnnyTig, 2022-05-20:
Hi @smagicld 

to your questions:

1. battery
https://belluna.eu/shop/100-ah-lithium-lifepo4-batterie-mit-bluetooth-funktion/

2. mosfet on/off
 as described in my first and second post yesterday, your iOS app is able the read everything. But I can imagine that your assumption is correct that you can't save anything or write anything in the bms with the app.
The buttons "Battery off" and "Battery on" have no effect.
Likewise, changes to the BMS data that the app pushes to the BMS are not saved the next time the data is read out.

3. Android app
Unfortunately, I do not own a device with Android. So I can't try out the bms.apk that the seller offers for download here https://belluna.eu/bt/.

ysmolski, 2023-11-11:
Same here, I cannot write config for almost the same battery (I think it is the same battery as yours but branded as Volt Polska). @JohnnyTig When you contacted the vendor, did they tell you if there is a PIN on this battery?

smagicld, 2023-11-11:
@ysmolski if you tell me your e-mail, I can provide you with an release where you can remove the battery pin.

desultorynz, 2023-11-27:
@smagicld can you also please offer me a version where I can "REMOVE" the Pin thanks, for Ios though.  I notice even when I enter the correct pin there's no button called "Clear Pin or Wipe Pin" which would be super handy, as simply re-setting the pin back to 000000 doesn't take the bms back to how it was before any pin was added into the bms, so having that option to clear or wipe the pin (even if it needs the password entered to do this) would be handy.