smagicld/xiaoxiangBMS #189: Advanced Features Unavailable With No Cellular Data

packrat1969, 2022-06-03:
I recently built a 4S LiFePO4 battery using a JBD-SP04S028A-L4S-150A BMS for my Camper/RV.  While boondocking out of cellular data range, I found it impossible to use the advanced configuration features of your app as it would only attempt to “phone home” for verification that I truly had purchased it.   When I selected the “config” screen, all I got was a “Restore Purchases” button, which of course didn’t work as I had no cellular data.  This it’s not the first time I’ve had to do this “Restore” after purchasing the app, but it was the first time that I had no recourse. This action should happen once, and then never again. It sort of defeats the whole point of paying for an app, if it has to verify its purchase status every freaking time.  Please fix this.

smagicld, 2022-06-03:
Hi
The in-app purchase is stored locally once verified. if you re-install the app - then you have to restore the purchases. without internet connection it is not possible - the status is hold / verified on apple servers.
there is no "phone home" - all calls are made to Apple servers to get the status.

packrat1969, 2022-06-03:
Hi,
Thanks for the fast response.  Yeah, when I installed the app the first time I purchased the upgraded version.  However, each time I went in to change the parameters in my BMS, the app asked me to “Restore Purchases”.  It wasn’t a problem until I was outside of my cellular coverage area.  This morning, I uninstalled and reinstalled the app, and then Restored Purchases again.  We’ll see if it stays put when I turn off all cellular and WiFi data.

On a totally separate note, is it possible to rename the various temperature sensors?  I have one on the BMS, one on the outside of the battery pack, and one buried in the middle of the pack.  It would be good to have customizable labels for each…

Sent from my iPhone and it’s tiny keyboard...

Richard