smagicld/xiaoxiangBMS #190: Text in Notification Fields is Cut Off

packrat1969, 2022-06-03:
The text in the attribute fields is cut off/incomplete in the Notifications section.  It looks like it’s left justified, when it should be centered in the field.  Email if pictures are needed for clarification.

smagicld, 2022-07-14:
Hi. can you paste me the picture. which version of the app? which iDevice are you using?
On my iPhones I was not able to reproduce it:
[image]

packrat1969, 2022-07-14:
Hello,
First off, thank you for your support of this app. What are your plans for upgrades/new features? Are you open to suggestions?

If you didn’t know this already, there does appear to be a following for this app and others like it on the DIY Solar Power Forum site (https://diysolarforum.com/).  I also think a vocal presence there would help you immensely in forwarding this product.

Currently I’m running iOS 14.8 on an iPhone 11. Since this is a work phone, and I have apps that rely on that version of the OS, I cannot upgrade it to the latest version of iOS 15.

Here’s a picture of what I see.

A couple of other issues. Is there documentation that describes how to use this feature? And if so where can I find it?

Is there any chance of renaming temperature sensors to indicate where they are located?

Sent from my iPhone and it’s tiny keyboard...

Richard

smagicld, 2022-07-14:
HI again

the screenshot is missing.

Which app do you mean, the XiaoXiang or BMS Tool?

Currently XiaoXiang is maintenance only. all new features and other BMS types are implemented into BMS Tool app.

"documentation that describes how to use this feature" - which feature? Notifications or something else?

smagicld, 2022-07-14:
regarding Temp sensor name - with next release you'll get pencil button next to the Temp sensor name - by pressing it you will be able to rename the sensor.

[image]

packrat1969, 2022-07-15:
Weird. Not sure why the picture didn’t get transmitted.  Trying again.

Sent from my iPhone and it’s tiny keyboard...

smagicld, 2022-07-15:
I still have no screenshot / can not understand/fix your issue.

building new release... but at least you'll get the option to rename temp sensors.

smagicld, 2022-07-21:
closing because missing feedback

closed 2022-07-21 by smagicld