smagicld/xiaoxiangBMS #191: Multiple Bluetooth BMS Monitoring

MattPaulin, 2022-06-04:
In marine installs, several drop in bluetooth only BMS batts are often used. The USA, NZ and Australia are brining in rules to require a visual and audio alarm to warn of an impending disconnect, for any reason (Over Volt, Under volt, temp etc). Without this it is not legal to use Lithium Batt tech in a boat.
Could this be included in future update - or a marine version of a multiple batt cable bluetooth BMS?

desultorynz, 2023-11-27:
@MattPaulin how do other drop in battery options give both a visual + audio alarm option for drop in lithium batteries?

smagicld, 2023-11-27:
to monitor multiple batteries you can use the newer version: BMS Tool. 
What would be your requirements for the monitoring? Whenever BT connections is dropped - send push notification?

Anyhow I'm not sure if any App will fulfil the legal requirements. iOS or Android can kill any app any time. in this case you will not get any notification. I think for this purpose dedicated HW is needed. 
But from the other side - this is what the BMS is for - to protect the battery (lithium) cells all the time.

MattPaulin, 2023-11-27:
Yeah, a notification if link is down would work. Most batts that can do this have CAN connections, and are expensive (Victron, Mastervolt).

MattPaulin, 2023-11-27:
> to monitor multiple batteries you can use the newer version: BMS Tool. What would be your requirements for the monitoring? Whenever BT connections is dropped - send push notification?
>
> Anyhow I'm [...]

So there is a new version that does multiple BMS's? Could I have a link or App name so I can find it? THX!

desultorynz, 2023-11-27:
> So there is a new version that does multiple BMS's? Could I have a link or App name so I can find it? THX!

It's called "BMS TOOL" on the App Store by the same publishing company who makes the existing Bluetooth app that your already using.