smagicld/xiaoxiangBMS #198: In app purchase issue

Afsheen70, 2022-07-28:
Hi,
I purchased your app. for 7 dollars and every time I attempt to go to Config. Screen, it asks me again to buy it. Can you fix this issue? It’s annoying 

thanks
Afshin Asharfy

smagicld, 2022-07-28:
Hi. 

Just delete the app completely and install again from AppStore. 

This should fix the problem permanently. 

If you still have this issue - let me know.

WhaleCaulk, 2023-05-22:
Hello, I have the same problem, I purchased the pro version of the app for $10 but when I tried using it the app told me I did not purchase. I tried restarting, deleting and redownloading the app aswell as contacting apple to try and solve it but nothing worked. How do I get the app to work?

smagicld, 2023-05-22:
If you go into Info -> InApp purchases and press restore purchases button, do you get the green check mark ✅ on the Pro Version?

WhaleCaulk, 2023-05-31:
No, if I remember correctly after I purchased it the app store said the
purchase was successful so I went back into the app but nothing showed up,
it still told me to buy the pro version. The restore button said there was
no purchase.

WhaleCaulk, 2023-05-31:
I did purchase the app from a newer phone just in case that was the problem
and it went through no problem. Although my mom who can see purchases said
to me that it got double charged

smagicld, 2023-05-31:
The purchase is linked to appleId. are you using the same appleId as you made the purchase? 

and double purchase is not possible with the same appleId - 2nd purchase will be always rejected.

so please evaluate the appleId you have been using for the purchase and make sure this is the one on your iDevice.

WhaleCaulk, 2023-05-31:
I did not use the same email as the first purchase. I'm not entirely sure
about the double charge but the two charges were the same at the same time
roughly.

WhaleCaulk, 2023-05-31:
> So this is the reason. you have to be logged on the iDevice with the email
> as purchased. if you were double charged - check it with apple. we do not
> have any evidence who (appleId) has made the purchases. Apple is managing
> all purchases.

Ok, thanks for the clarification, I will do that.

ulihuber, 2023-06-11:
I hab the same issue. I deleted  the app accidentally and reinstalled from apple appstore.
Now it asks me every time for ‘in App-Käufe wiederherstellen’. I get the green checkmark for ‘pro’ but only within one session.
Can this be caused by a change of my email address after leaving my company? How can I fix it?

smagicld, 2023-06-11:
Hi @ulihuber If you get the green checkmark - it means you bought it. please delete once again the app (with all data) and reinstall from App Store. it should be then ok. let me know.

smagicld, 2023-06-13:
Hi Uli

kannst du mir ein screenrecording machen - wo der inApp restore + kill der App und noch ein Mal das inApp Bereich sichtbar ist? So kann ich besser nachvollziehen. Danke