smagicld/xiaoxiangBMS #199: Add required 100, 90, 70, 50, 30, 10% fields

Marvo2011, 2022-07-29:
Hello,

please add the for SOC calculation required 100, 90, 70, 50, 30, 10% voltage fields.
[image]

https://diysolarforum.com/threads/jbd-bms-how-to-correct-inaccurate-soc-value-%E2%9C%85.40399/

smagicld, 2022-07-29:
Thank you for the input. It looks this are newly introduced fields for new FW. I'll check it and try to integrate into the app!

Marvo2011, 2022-07-30:
Thanks I think here are some interesting parameters:
https://github.com/FurTrader/OverkillSolarBMS/blob/master/Comm_Protocol_Documentation/Capacity_parameter_registers.md

smagicld, 2022-08-02:
Hi Marvo

first beta is there. 
have to test it with older firmwares and then it can be released. 
mid you drop me your email - I can add you into the test user group so you can have it before official release. 

[image]

Marvo2011, 2022-08-02:
Thanks, I have send you the mail to your company / website mail.

smagicld, 2022-08-02:
Got it. Please note it can take up to 2 days till Apple releases the beta version (you will receive TestFlight email)

RayOei, 2022-08-16:
Although not _directly_ related but as the RN state: it _may_ work on firmware 20: is there information available somewhere on how to upgrade the BMS to JBD firmware v21?

Marvo2011, 2022-08-16:
You can upgrade the firmware with the JDBTools for Windows, but you have to get the right firmware file from your distributor.

Arbitaly, 2023-03-15:
Hello, the update that appeared in update 1.2.36 does not work correctly. As you can see in the screenshots, I have the battery voltage set in% for each level, but the charge is shown based on the calculation of the spent capacity (calculated through the consumption of amperes). But if such consumption is considered erroneous, then the% of the remaining charge is shown incorrectly. Is it possible to make a % charge calculation based solely on battery voltage?
[image]
[image]

smagicld, 2023-03-15:
Hi. Your problem is not related to the update nor the app. 
the SOC is calculated and reported by the BMS. It measures the energy in/out and based on this the soc is calculated. 

SOC based on cell voltage (first issue: based on which cell? - if you have 12? Or on cell volt average? ) then the cell voltage strongly depends on the current. So SOC based on voltage is not the best way to calculate the SOC.

Arbitaly, 2023-03-15:
Is it possible to add such a way of tracking based on voltage? perhaps on a separate tick? because I (and many) check the charge level only in parking lots, when there is no load. And measurement by consumption always turns out to be erroneous in one direction or another, unfortunately. Please. I bought access to the IOS app. :)

Arbitaly, 2023-03-20:
> SOC based on cell voltage (first issue: based on which cell? - if you have 12? Or on cell volt average? ) then the cell voltage strongly depends on the current. So SOC based on voltage is not the best way to calculate the SOC.

My battery is 10 cells, % of charge can be calculated by 1 cell, which can be selected or by the average value of all cells, the number of cells is indicated in the parameters, so you can sum up the voltage of all cells to calculate the average value.

Please, could you add this feature? at least on a separate point (perhaps with a warning that this is not the best way and it will not show correctly under load). But for me it would be perfect.