smagicld/xiaoxiangBMS #202: My paid version disappeared after update to 1.2.36

marc-hugo, 2022-08-16:
Hi,

I have a paid version of XiaoXiangBMS, but after the latest automatic update on iPad to version 1.2.36, I can't see the pages with settings anymore. How can I get my paid version back?

(as my BMS has a secret Liontron password, I can't open the Config-button in the app, so I cannot revive the pro mode).

Regards,

Marc

smagicld, 2022-08-16:
Hi. Info Icon on main screen and then restore purchases

marc-hugo, 2022-08-18:
Hello,

Check button for Pro version is green, but the button Restore Purchases 
is dead. What should I do next?

Regards,

Marc.

smagicld, 2022-08-18:
Hi

Is your internet connection working?
Are In-App purchases enabled in iOS Device Settings?

marc-hugo, 2022-08-18:
1) yes
2) yes

smagicld, 2022-08-18:
Can you attach your screenshot please.

The Resort purchases Button is grayed out only when purchases were restored or the verification is ongoing (see my screenshot).

have you tried to kill and start the app again?

[image]

marc-hugo, 2022-08-18:
1) yes
2) yes

See attached screen shot: the button Restore Purchases is grayed out.

smagicld, 2022-08-18:
there is no screenshot.
otherwise try to delete the app and re-install from App Store again, but I bet it is related to your iOS device settings (probably in-app purchases).

closed 2023-01-10 by smagicld