smagicld/xiaoxiangBMS #204: What Information is needed to add BMS to your App?

tomtom1204, 2022-09-13:
Hi,

I use your paid version iOS App with my Liontron and it works fine. But I own another LifePo Battery which your app discovers but is unable to read. I added two pictures (2 as I don't know what information is essential and not all fits on my screen) Information from a bluetooth scanner app, will you be able to deduct the necessary changes from this?
[image]
[image]

smagicld, 2022-09-13:
Looks to me like WATTSTUNDE battery. 

Try the newer app: BMS tool.

tomtom1204, 2022-09-13:
I recently downloaded the BMS Tool (App version 2.0.12) and it also discovers the battery but could not readout any parameters.

tomtom1204, 2022-09-13:
The label on the battery reads "JWS Solar".

smagicld, 2022-09-13:
it looks like it is completely different BMS using same Bluetooth UUIDs.

i could try to implement it, but need the hardware for reverse engineering. 

So If you want it, you have to sponsor the HW.

tomtom1204, 2022-09-14:
OK, thank you for the offer but as the Battery would be destroyed afterwards there is no meaning in this. I will be using the Android App from JWS instead.

closed 2023-01-10 by smagicld