smagicld/xiaoxiangBMS #205: Can't find my battery after changing the HW specific bluetooth name. Was only trying to change the name of my battery

DarthHader, 2022-09-14:


DarthHader, 2022-10-01:
> 

Anybody out there?

smagicld, 2022-10-01:
Most probably other device / smartphone is already connected to the BMS. 

Is the led on BT dongle on?
Have you tried to re-connect the BT dongle with BMS board?
Is the battery charged?

DarthHader, 2022-10-01:
Unfortunately the battery is sealed. I brought it pre-built. I was messing around with the app and was only trying to change the name. Instead of the alias name I changed the HW specific name. I'm wondering if there is any way of resetting the bms without opening it. Like jumping the positive and negative to force a shut down? As I can't find the battery after changing the HW specific name(whatever that means) I'm pretty sure that the bms is lost to bluetooth discovery without cutting the battery open but I'm open to ideas. Thanks for responding!