smagicld/xiaoxiangBMS #206: Will no longer turn Battery On or Off or turn Charge or Discharge On or Off

TreyMagnus, 2022-09-28:
I have several JBD BMS's from 4s, 7s, 14s and 16s and they were all purchased over the past 3 years.  I do not know the FIRMWARE versions of them all but this weekend I found that on some of them, I could no longer turn the Battery ON or OFF, or the  Charging, or Discharging Ports ON or OFF.

I have also heard reports of not being able to update parameters but have not yet tried that myself.

I am trying to find firmware files to update my BMS's (using JBDTools) with the newest Firmware but no one is cooperating so far in providing the Firmware update files (or specific instructions on how to apply them so we do not brick our BMS's).

In the meantime, I am requesting that you put an older version of the app back onto the Apple App store so we can downgrade to the older version until either we can upgrade our BMS Firmware or you are able to get backward compatibility with all Firmware versions.

smagicld, 2022-09-29:
If you can not switch on/off the port - the to 99% your BMS is protected by PIN (and this is not stored in the app). 

the port can be switched on/off only if configured mode can be entered. 

try following: connect to BMS, go to BMS settings (config). You should be asked for hardware pin. Enter it (and make sure the checkbox to store it is checked). Go back and try to switch the port on or off (a warning message / pin query for this specific situation is in the backlog)

TreyMagnus, 2022-09-29:
That cannot be the problem as I have NEVER entered a PIN for ANY of my
BMS's.
I have only noticed that option recently and (on at least some of them) it
says that option is not supported by the current firmware.
I have other BMS's that can be turned on and off (both Battery and Charge &
discharge ports individually) from the main screen (not in configuration
mode).
I will look at the different firmware versions to try and see if there is a
pattern of which one(s) works and which one(s) do not.

TreyMagnus, 2023-03-08:
I am still having this problem from time to time (like today!).  I have added PIN number protection to all of my BMS's.  I do not have problems with the PIN number being saved in my iphone.  I AM able to go into configuration and make changes and save them but it will not turn the charging ports on or off individually or together by the Batt On / Off buttons.  It just keeps saying "Port Change Failed".

Christmas12345, 2023-03-08:
Same here, the BMS has “lost” the manufacturers name so I think I will need
to reprogram it with the UART tool Bluetooth won’t work. BlueCap is the
same.
Regards Darryl.

smagicld, 2023-03-08:
> I am still having this problem from time to time (like today!). I have added PIN number protection to all of my BMS's. I do not have problems with the PIN number being saved in my iPhone. I AM able to go into configuration and make changes and save them but it will not turn the charging ports on or off individually or together by the Batt On / Off buttons. It just keeps saying "Port Change Failed".
> This is a DP04S007L4S150A
> Manufactured Aug. 19th, 2022
> Firmware Version 23

Can you provide me the logs? I would like to see what is the BMS response on Mosfet On/Off command.

Christmas12345, 2023-03-08:
A can’t get past the initial report screen. Configure doesn’t allow me to
get in.
Regards Darryl.

Christmas12345, 2023-03-09:
> I was asking TreyMagnus for the logs.
>
> In your case it looks like the bms is not powered. Check the wiring.

Not the case. The BMS has lost part of its brain. This only started
happening AFTER the manufacturers name disappeared. 🤷🏽‍♂️🤷🏽‍♂️🤷🏽‍♂️

Christmas12345, 2023-03-09:
So sorry for interrupting.

Christmas12345, 2023-03-09:
I can’t understand how Bluetooth still works if the BMS has no power.
🤷🏽‍♂️

smagicld, 2023-03-09:
Because it is separate low power board. The bms itself takes the power from sense lines + B-  as ground. If not correctly connected it can happen that it will not work. 

It can also be the connection between BT board and BmS itself. It is hard to diagnose remotely. 

You could also try to disconnect the main sense connector from BMS for 10 minutes to reset the BMS.

Christmas12345, 2023-03-09:
> I will send it on. I’m not able to, or capable or confident enough to do
> it myself. 😖

So good to know. Thanks again. 🙏🏾

TreyMagnus, 2023-03-09:
Where do you find the logs in the app?

closed 2023-03-19 by smagicld

Christmas12345, 2023-03-19:
Cheers, thanks for the information.

TreyMagnus, 2023-03-21:
What is the solution?