smagicld/xiaoxiangBMS #208: Bluetooth EEPROM-Read fails ‚… no confirmation from BMS, try again‘

dadaludwig, 2022-12-03:
Bluetooth EEPROM-Read fails (see error message attached).
Always worked until now.
I have four 13S9P battery stacks with Xiaoxiang Bluetooth-BMS. Behaves same with all.
USB serielleres PC works. Could read and write all.
Please help. 
[image]

smagicld, 2022-12-03:
Please provide exact BMS Type and firmware version.

dadaludwig, 2022-12-03:
Thanks for quick response. Attached a screenshot.
Regards
[image: PNG-Bild]
BMSType_Firmware.png
<https://drive.google.com/file/d/1xE_tYp9OA1TrGt3eAy0YFnGYHAYqmPln/view?usp=drivesdk>

dadaludwig, 2022-12-03:
[image]

smagicld, 2022-12-03:
> Thanks for quick response. Attached a screenshot. Regards [image: PNG-Bild] BMSType_Firmware.png <https://drive.google.com/file/d/1xE_tYp9OA1TrGt3eAy0YFnGYHAYqmPln/view?usp=drivesdk> Magic ***@***.***> schrieb am Sa. 3. Dez. 2022 um 20:37:
> […](#)

Who is the manufacturer? From the screenshot I can’t identify exact hardware model. 
is this Jiabaida? (JBD)

dadaludwig, 2022-12-03:
> Thanks for quick response. Attached a screenshot. Regards [image:
> PNG-Bild] BMSType_Firmware.png

BMS Name is 15660064-SP15S020-P13S-60A-60AH. Does this help?

smagicld, 2022-12-03:
Yes. I order to analyse your issue I have to get similar BMS with same FW. Can take some time. Will keep you updated.

dadaludwig, 2022-12-04:
> OldApk1.png
>
> <https://drive.google.com/file/d/1xX_sLY8soCXHSLdkGVUMqD6eqe0_cxNs/view?usp=drivesdk>
> [image: JPEG-Bild]
> OldApk2.png
> <https://drive.google.com/file/d/1RMmdbMs7geSYdphHKZSqyoZfCSQo9p24>
> [image: JPEG-Bild]
> NewBMSAppStore.png
> <https://drive.google.com/file/d/1Je4SJICLkHQnKdwYRfQbvmhMNF2Jwk-3>
> Great news, thanks.
>
> I’m using iPhone-App. Can you provide the newest Android -APP?
> I installed a old version on my sons S10 yesterday
> (xiaoxiangprotection-2.1.1022.apk), and it’s crap (see attached screenshot).
>
> And what about the new ‚BMSTool‘ in App Store? (See screenshot). Looks
> very promising. Still under construction or do you have a beta version I
> could try to help us narrow down the fault? (If this one should also fail
> the EE-PROM read, which I don’t believe, then maybe it’s not the app)
>
> Regards
> Dieter

[image: JPEG-Bild]

dadaludwig, 2022-12-04:
… Sorry, I Mixed the file names. OldApk1 should be NewBMSAppStore and visa
versa.

dadaludwig, 2022-12-04:
.. and short video to demonstrate the error message

smagicld, 2022-12-04:
und wo finde ich das Video?
Du schreibst der Readout hat bis jetzt immer funktioniert - jetzt bricht es ab. 
es kann sein das du zu weit entfernt bis von dem BMS oder etwas stört das Signal.
Kannst du ganz nah an das BMS / Bluetooth Dongle und dann auslesen?

dadaludwig, 2022-12-04:
… cool, du sprichst deutsch 😁🍺. Wie gesagt, hat immer problemlos
geklappt. Ging auch meistens viel schneller mit dem blauen Balken. Im Video
sieht man, dass es nix mit der Entfernung zu tun hat.
Hast du die Android-App? Dann probiere ich es mal darüber
(Ausschlussprinzip). An meinen iPhone kann es nicht liegen denn ich habe es
mit meinem iPad auch probiert (im Video zu sehen).

VG

dadaludwig, 2022-12-08:
… hast Du ein Update für mich? Habe nix mehr von Dir gehört … was ist mit
der Android app? Denn Weg könnten wir mal ausprobieren …

smagicld, 2022-12-08:
Hi
Du schreibst es hat alles mal gut funktioniert. 
Es Hatcher schon lange kein App Update gegeben - somit kann es nicht an der App liegen. 

Du schreibst das gleiche Problem hast du mit dem iPad - für mich ist das eine Indiz es liegt an der Batterie / BMS - sehr wahrscheinlich ist diese zu weit oder das BT Signal wird gestört. 

Ich kann diese Probleme nicht reproduzieren. 

Link an dein Video fehlt. 

Und was ist mit der Android App?

dadaludwig, 2022-12-08:
... ich habe das video hier nochmal angefügt (zum dritten mal, hoffentlich
klappt es diesmal)

Zum einen siehst Du, dass ich ganz nah dran bin. Zum anderen kann es nicht
an den BMS liegen denn habe insgesamt VIER stacks (im Video drei und einen
anderen Pack der nicht im Video zu sehen ist) und ALLE haben das Problem.
Ja Du hast recht, das Verhalten (blauer Balken geht viel zu langsam hoch)
deutet auf 'nicht nah genug dran' oder 'Unterbrechung' hin, ich kenne das
Verhalten, aber wie Du im Video siehst, kann man das als Fehlerursache
ausschließen.
Es muss an der App liegen (am Iphone kann es nicht liegen denn über mein
iPad, im Video zu sehen, funktioniert es auch nicht), daher noch einmal die
Frage: Ihr habt doch auch eine App für Android-Geräte, oder nicht? Wenn ja,
könnt ihr mir die zur Verfügung stellen? Wenns mit Android auch nicht
klappt, dann wird's muss es wohl am BMS (oder eher am Bluetooth-Interface)
liegen was aber dann 'ein sechser im Lotto' wäre (extrem unwahrscheinlich
aber wahr).

P.S. wenn das video wieder nicht zu öffnen ist dann hast Du vielleicht
einen anderen Tip, wie ich Dir die File zukommen lassen kann. File name
ist: BMS-NoResponse.mov (2.642 KB)

VG
dieter

dadaludwig, 2022-12-08:
> Hi Dieter
>
>    1. Das Video kann ich immer noch nicht öffnen. Wo hast du es
>    angehängt? oder sende mir ein iCloud link.
>    2. Wir bieten keine Android App
>    3. Wie schon erwähnt - die App wurde nicht upgedated - es kann nicht
>    an der App liegen. Vor mir habe ich 10 verschiedene BMS - kann alle ohne
>    Probleme auslesen / kann dein Problem beim besten Willen nicht
>    reproduzieren.
>    4. Das Problem mit Auslesen deute ganz klar auf Verbindungsprobleme.
>    Da es nicht auf dem iPad geht - würde bestätigen das es irgendwelche
>    Interferenzen gibt oder die BT Dongles zu stark abgeschirmt sind.
>    5. Was du noch machen kannst: gehe ins About Screen und drücke auf
>    Version Nummer. Dies schaltet das Debugging ein. Wenn eingeschaltet - gehe
>    ins BMS und versuche es auszulesen bis der Fehler kommt. Danach zurück ins
>    About screen und drücke noch ein mal auf die Version. Sende das E-Mail mit
>    den Debug Loggs.

.. danke für den Tip mit dem About screen. Probiere ich mal und melde mich.

Ich habe das video von .mov auf .mp4 umgewandelt und angefügt. Kannst Du's
jetzt öffnen? wenn nicht dann plage ich mich weiter mit iCloud :-)

VG

dadaludwig, 2022-12-08:
.. übrigens, hast Du eine andere App als ich? Wo bitteschön soll ich auf
version tippen? Die Versionen sind alle ausgegraut (Bild1). Da geht nix mit
tippen.
 [image: Bild1.jpg]

Ich ich da überhaupt richtig. ich bin ganz normal über die Startmaske
(Bild2) rechts unten auf "info" gegangen.
[image: Bild2.jpg]

onlineedition, 2022-12-11:
I have exact same issue after updated Xiaoxiang iOS app to v1.2.36.  My BMS is on v25 firmware.  It Worked fine on my another iso phone with previous Xiaoxiang app v1.2.35.  I think the issue with v1.2.36 with the new extra battery percentage 10% 30% 50% 70% 90% and because the BMS does not have these setting to begin with.  The Bluetooth trying reading it and timeout because these field are empty data.  I’m also trying to created a new profiles with all the setting and fail to write it.  I think my BMS need correct different firmware than v25 that I currently  on.  I be waiting for the solution.  Meanwhile I’m working on downgrade the firmware from jbd to older v21. Or Somehow need delete iso v1.2.36 and install with the previous v1.2.35.

smagicld, 2022-12-11:
The problem is identified. This is because there is some BMS FW outside that should support this fields but does not so when the App tries to read then it comes into timeout. There is a test build out now, should be released next week / fixed.

smagicld, 2022-12-11:
But it‘s just an warning - you can still set other parameters.

onlineedition, 2022-12-12:
> The problem is identified. This is because there is some BMS FW outside that should support this fields but does not so when the App tries to read then it comes into timeout. There is a test build out now, should be released next week / fixed.

Sweet, thank you so much I be waiting for it.  I’m still able to monitoring two of the BMS that I have with the other phone with older version.  Thanks.

smagicld, 2022-12-12:
If you send me your email to  Xiaoxiang@e77code.com I will add you to the test user group.

smagicld, 2023-01-10:
Fixed in 1.2.37

closed 2023-01-10 by smagicld