smagicld/xiaoxiangBMS #209: Pro Version not available anymore

stephan2403, 2022-12-06:
Hi,

hi I bought the pro Version last year vor 9,99€. Since that I always had to restore the purchase when I want to enter settings or disable the ports. Annoying but ok. But now the app said there is no purchase to restore. Already tried reinstallation. Didn’t help. What can I do? Don’t want to sell 10€ again. Thanks!

smagicld, 2022-12-06:
Hi

the in app purchases are bound to appleId. All your purchases must be under the appleId available and this is what the app is checking. 
If you did not changed your appleId and it does not work - you have to ask Apple.

closed 2022-12-06 by smagicld

stephan2403, 2022-12-06:
Hi, I've checked my iCloud. Purchase is in the list but app don't work. How to link both?

smagicld, 2022-12-06:
Hi
The question is if you're still using the same AppleId for AppStore?
Anyhow, what is the message if you go within the XiaoXiangBMS App into Info -> InApp Purchases and press the button "Restore purchases"?

stephan2403, 2022-12-06:
Hi,

Yes everything is the same. AppleID Smartphone. Nothing changed.
Today i also had contact to apple support: A nice lady checked everything. The purchase is in my list, but she said the pro version expired and can’t change it because it is a third party app. How could the pro status expire since it is a one time payment for lifetime. Isn’t it? Have there any changes been made? I’ve paid 6,99€. Now its 9,99€.

The message shown in the app was: there are no purchase date to restore.

BUT GOOD NEWS: i checked again this evening and it worked. no changes, just because i want to see the messagings again to send you an answer. 

So hope this helps you to avoid this issue in future. now we can really close the case :-)

Thanks!
Stephan

stephan2403, 2022-12-06:
> Hi Stephan
> As I wrote before - we're are not managing purchases - it's Apple who is telling the app if something was bought or not.
> Also the pro Version has no expiration date - the nice lady gave you wrong information.
> It can imagine apple had some issues and after your call they fixed it.
> finally I'm happy you could restore your purchases!
> Kind regards
> Lukas

Ok cool!Now it’s working. Also thanks a lot for your swift respond!!Stephan