smagicld/xiaoxiangBMS #213: List Bluetooth signal on BMS

adoyuran, 2023-01-02:
I am using overkill solar 4S BMS with Bluetooth dongle. I lost the signal. What could be wrong?

smagicld, 2023-01-03:
What do you mean with “lost signal”?
If the BMS dongle is not visible - make sure no other app / device is connected to the BMS over Bt (the Led on BT dongle is on). 

Sometimes the BMS hangs - you could try to disconnect voltage sense connector for 1 min and connect it again.

adoyuran, 2023-01-03:
Normally when connected single turn blue.there is no more light. Do you
think it is dead?

Dwightholt, 2023-01-25:
I am having the same issue. I have lost Bluetooth connection with my BMS. If I disconnect power from my BMS and then reconnect it is visible for a few seconds in the device list then I goes away. If within that few seconds I pull power then reconnect it I can establish the connection to my BMS. I disconnect from the app and restart app the BMS will not show up on the device list until I pull power again and quickly tap to connect, otherwise the connection will disappear from the list again. It appears that the BMS is working but the Bluetooth is losing its connection. The blue light on the Bluetooth device is on solid blue as well, with a blinking blue light in the BMS. I assume this is normal. 

Any ideas how to fix this issue, or should the Bluetooth be replaced? Please help.

smagicld, 2023-01-25:
The blue led on BT dongle means someone is already connected. If you‘re not the one who is connected - then probably someone else is connecting to your bms. It could be also you have 2nd device with running app/ connected to bms or another app like the BMS tool which is connecting to the bms.

Dwightholt, 2023-01-25:
Thank you. I was able to find one of my IPads and it was connected to the BMS. I shut it down and was able to connect with my IPhone as normal. I had no idea that the iPad made a connection. Th sinks again for your help.

closed 2023-03-08 by smagicld