smagicld/xiaoxiangBMS #215: App (pro) dont work

Sysko75, 2023-01-04:
Ho, i use till one year Xiaoxiang app (i buy the pro version) now config button dont work and Also battery on  and battery off … how i resolve? 
bms Jbd firmware 22
Inside a ULTIMATRON france batteri 4S
App vers 1.2.37

smagicld, 2023-01-04:
Hi

please provide more details / screenshots. What does not work exactly? Please provide steps to reproduce your issue.

Sysko75, 2023-01-04:
[image]

Sysko75, 2023-01-04:
All in red signed dont work!!! If i tap on config dont go in configurazione mode

smagicld, 2023-01-04:
More details please. What happens if you press config button? Do you get any error message? If not, what happens? Which app version are you using? Without more details it is hard to help you.

Sysko75, 2023-01-04:
If i press config i dont get any error but nothing heppen! App version i write in my first message is 1.2.37

smagicld, 2023-01-05:
I'm not able to reproduce your issue on any of our test BMS or iOS Devices.
I will need debug logs.
Please go into About screen and tap on the Version / Build number. This will enable Debug Logging. then go back, connect to bms and press the config button. 
next: go back into About screen and tap again on the Version / Build number. Send me the debug logs.

smagicld, 2023-01-10:
Thanks for the 1 star rating and not providing more logs.

closed 2023-01-10 by smagicld

Sysko75, 2023-01-10:
What for rating?

Inviato da iPhone