smagicld/xiaoxiangBMS #216: Lost connection to the BMS.

Christmas12345, 2023-01-09:
The Bluetooth still works but the BMS has lost its identifier so can’t be switched on. Hate not being able to fix something that appears so simple. Could be as simple as UART reprogramming but I don’t have a pc. nRFconnect works but doesn’t allow me access to reset the BMS. Is there a way

Christmas12345, 2023-01-09:
Rowe.Darryl@gmail.com

nirgo, 2023-07-12:
> The Bluetooth still works but the BMS has lost its identifier so can’t be switched on. Hate not being able to fix something that appears so simple. Could be as simple as UART reprogramming but I don’t have a pc. nRFconnect works but doesn’t allow me access to reset the BMS. Is there a way 

Did you find a solution? Seems I'm in a similar boat.

Christmas12345, 2023-07-12:
I’m told the main board has lost it’s connection. Mine was a physical
problem (loose plug). Good luck.
Cheers mate.