smagicld/xiaoxiangBMS #218: JBD BMS config pin not working on Apple I Phone

Dwightholt, 2023-01-16:
I can not get into my BMS any longer because is keeps asking for the security code. I put in 6 0s as is the default but it just keeps looping back to put it in again. I also can not turn off or one the charge or discharge ports any mor as well. There seems to have been a change and now the app will not work. I deleted the app the reinstalled to see if that would work but I’m shut off with out being able to make changes. What’s the fix???????

smagicld, 2023-01-16:
if the hw pin is set then you are not able to turn the ports on or off. 
Are you sure you did not set the hw code?
Or is your BMS “visible” to others? Maybe someone else set the code for you?

Dwightholt, 2023-01-16:
I did not set a pin. It has always been left to default and the app says it did not use a pin. It was working just fine today then it started to ask for the pin. I went to my other battery pack, which is about an hour away, and it did the same thing. I know there was an update to the app and the FW version of the BMS. My BMS FW is 20, so how do I update this and it seems that the app update only works with the new BMS updated FW.

Dwightholt, 2023-01-16:
[image]

Dwightholt, 2023-01-16:
I have three battery’s in my boat and there all the same.

smagicld, 2023-01-16:
I will try to reproduce it on BMS with FW 20 (have to check if I have one). But can you exclude that some other persons set Pin on your batteries? Because I’m the App was no code change in this area.

Dwightholt, 2023-01-16:
For sure no one changed the PIN code. They would have to change it on three batteries there then drive to my house 1 hour away and change it too.

floreno, 2023-01-16:
Had the same issue two times with two batteries. Got them in low charge at 0% and could access the settings without pin, put it on the charger and after some hours trying to access the settings again, it keeps asking for the pin (its not 666666, 000000, 123456). There is no one else in normal range of the bluetooth-device. only someone with "hitec-antenna-(array-)system" could connect and set a pin without knowing me.

smagicld, 2023-01-16:
HI @floreno @Dwightholt 
I've tried to reproduce your issues, but was not able to do so. But I do not have any BMS with FW version 20 (11, 19, 21 and 41 tested).

Can you please send me some debug logs?
1. go to main screen -> tap Info button -> tap the Version Number within Logo.
2. Info Alert will popup confirming debug mode is on.
3. connect to your BMS
4. try to toggle input port
5. try to go into configuration. enter PIN 333333 (so I can identify it in the logs). 
6. go back to main screen
7. tap Info button -> tap the Version Number within Logo
8. send me the logs

I hope to find some details there.

Dwightholt, 2023-01-16:
Just got my Apple iPad out and it works. I did not install any updates from Apple on my iPad but my phone did get a current update.

Dwightholt, 2023-01-16:
I got back into my BMS via my in updated IPad and told the BMS HW pin to change to default 000000 then I went to my updated iPhone and was now able to get into my BMS. So it seems that the issue might be the software update to Apple. Maybe it screwed up with the app and wiped off the pin.

Dwightholt, 2023-01-16:
When I get back on my boat I will try the same there

smagicld, 2023-01-17:
I think I found the bug with config section on some specific FW version. I’ll prepare Test build and release it asap.

Dwightholt, 2023-01-17:
Thank you I look forward to getting the new released version. Please let me know when it’s out and I will delete old version and install new.

smagicld, 2023-01-18:
Just as Update: 
I've spent whole day analysing your logs and trying to reproduce your BMS responses that are set while entering the PIN or trying to toggle charging/discharging code. 
The Response codes says the Hardware PIN is set and it does not allow to toggle ports or change configuration.

So if it was not you - then someone else has set the PIN for you on your BMS.

I will upload today new version (v1.3.0) with special button to reset the hardware pin.

Once you have it - just press the button (Remove BMS Password) and the HW password should be cleared.

[image]

Dwightholt, 2023-01-18:
Like I had said earlier, it seemed to happen after I updated my phone. My iPad was not updated and I was able to get in. No one changed the code.

Thank you,

Dwight Holt
***@***.***
253-677-4554

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F32357732%2F213038353-45dc23cd-6c80-4f52-99ce-f8d8ce111914.png&data=05%7C01%7C%7Cecc68831f31c4588d79a08daf8e700e0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638095968284613141%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=sdJNqEshG0b0dEXg%2BOPjPegSodZea4D4PZapNnjUnTc%3D&reserved=0>

smagicld, 2023-01-18:
But then it explained everything: you must have set the PIN some time ago and forgot about it.
and because once successfully authenticated the PIN is stored in the App.

Because you've your phone the settings were gone - so there was no pin stored in the App and you have to enter the valid one.

And it works on your iPad because the PIN you entered once is stored there.

So for you the fastest solution is to connect with the iPad, go into config and set new PIN. then enter the same pin in the app on the new iPhone.

Dwightholt, 2023-01-18:
So why is it that a few min earlier I was in the BMS making some changes and then it would no longer work. Then I drove one hour to home and it would not work on my battery system at home, one hour away from where I first started having issues. The pin was never changed, something in the app went off track and no longer worked.

Thank you,

Dwight Holt
***@***.***
253-677-4554

smagicld, 2023-01-18:
If you deleted or reinstalled the app or installed on new phone - then the local PIN store within the app is empty. so if you try to connect to any bms where you have set PIN before - you'll have to re-enter it.

floreno, 2023-01-20:
Did not reinstall or anything with the app. But it could be that i had set a pin but set it to store it. Question is why the app lost the pin 2 times on different bms. After first pin „was lost“ I could access the second bms till that pin got lost too.
Shouldn’t the pins stored in iOS keychain?

floreno, 2023-01-23:
Disassembled my battery and found a note i inserted there with the correct pin. With that pin i was able to connect the app (1.2.37). But why did the app lost the pin?

Klettermobil, 2023-01-23:
Unfortunately, I can not turn on the battery after the software update 1.2.37.
 The switch for the charging port also no longer responds. Is there a solution soon ?
Does it make sense and is it possible to update the firmware version 20 ?

smagicld, 2023-01-23:
> Did not reinstall or anything with the app. But it could be that i had set a pin but set it to store it. Question is why the app lost the pin 2 times on different bms. After first pin „was lost“ I could access the second bms till that pin got lost too.
> Shouldn’t the pins stored in iOS keychain? BTW BMS-Firmware version yesterday displayed 25 and today it’s 28 WTF?

The App just displays what the BMS is reporting and it has no function to update FW. No Idea how it can happen....

smagicld, 2023-01-23:
> Unfortunately, I can not turn on the battery after the software update 1.2.37.
> The switch for the charging port also no longer responds. Is there a solution soon ?
> Does it make sense and is it possible to update the firmware version 20 ?

port enable/disable is not possible when the bms is protected by pin. it returns then an error code / failed command.
I can check it in the logs, just send me then:
1. go to main screen -> tap Info button -> tap the Version Number within Logo.
2. Info Alert will popup confirming debug mode is on.
3. connect to your BMS
4. try to toggle input port
5. go back to main screen
6. tap Info button -> tap the Version Number within Logo
7. send me the logs

Klettermobil, 2023-01-23:
Thank you, I will try!
My pin works, this should not be the problem.

smagicld, 2023-01-24:
HI Tom

Das Problem ist, du kannst die Ports nicht EIN/Ausschalten weil den BMS lässt sich nicht in Konfiguration Modus setzen. Das weil er mot PIN geschützt ist:

[ℹ️] [2023-01-24 09:12:31.5260] >>> sent bytes to BMS: [0xDD, 0x5A, 0x00, 0x02, 0x56, 0x78, 0xFF, 0x30, 0x77]. Command: FACTORY_MODE
[💬] [2023-01-24 09:12:31.6190] <<< received bytes from BMS: [0xDD, 0x00, **0x82,** 0x00, 0xFF, 0x7E, 0x77]. Command: FACTORY_MODE
[‼️] [2023-01-24 09:12:31.6610]  decodeBinaryBMSdata  - received failed response: 0xDD, 0x00, **0x82**, 0x00, 0xFF, 0x7E, 0x77

Kennst du den PIN? Fall je, gehe in BMS config. Du wirst nach pin gefragt. Gebe dein PIN ein. Dann kannst du zurück vom Config screen und Port ein/ausschalten.

Klettermobil, 2023-01-25:
Hey Du …Danke für deine Mühe.Ich habe meinen PIN.Ich kann meine Daten auslesen.Ich kann einen neuen PIN vergeben.Aber den Lade Port schalten kann ich nicht. Fehlermeldung, weiße Schrift auf grau: „Fehler beim Port umschalten“Ich hänge dir mal Screenshots dran.Danke 🙏 Lg Tom0163.7321465

smagicld, 2023-03-08:
Es gab ein Bug wo der PIN nicht gespeichert wurde. mit 1.3.0 ist es gefixed (also einmal in Konfiguration Bereich und dann zurück)

closed 2023-03-08 by smagicld