smagicld/xiaoxiangBMS #219: iOS app constantly loses PIN

floreno, 2023-01-25:
It’s now the third time the app lost the saved pin. While I’m writing i checked again and the pin was back there (no pin input request) but I guess early or later it’s lost again. I have not updated the app or the iOS system.
It would be nice to store the pin in the secured iOS keaychain.

smagicld, 2023-01-25:
please describe steps necessary how to reproduce. 
the local pin can only get lost if the app was deinstalled.

floreno, 2023-01-25:
I definitely did not reinstall app. Last loss, I just inserted the pin again, after half an hour I tried again, and app keeps asking me for pin. After I restarted the app the pin was back there. 1h later it keeps asking again for pin and restart app don’t change anything. It seems to me the app is somehow lagging while accessing the pin-protected values like manufacturer. I’ll will check out the next days if I can reproduce it.

floreno, 2023-02-04:
Did today a check with two different bms and switching between them seems to constantly reproduce the pin loss. I have send the logs to u.

smagicld, 2023-02-04:
And I responded to you immediately:
"I’ve seen you’ve tried to enter 3 different PIN’s but all of them were rejected by the bms.
So none of them is valid.

And what do you mean exactly it is loosing PIN?
Describe how to reproduce."

For now I see only you are trying with wrong PINs: 176983, 498146, 333333.

So please describe the steps and attach screen recording with reproduction steps.

sarah1331sarah, 2023-03-04:
I have also had this several times recently I only have one BMS and one password I tried another password (wrong) then the correct one hopefully it “remembers” this one

sarah1331sarah, 2023-03-04:
Actually just happened again and sent in the logs

smagicld, 2023-03-04:
They issue is know. The pin is not stored when store pin checkbox is checked. Fix is already on the way, should be released in 2-3 days.

sarah1331sarah, 2023-03-04:
Thank you so much I was only trying to help

smagicld, 2023-03-08:
Hi
fixed in 1.3.0 version (released by apple now). Please let me know if you experience still issues with the pin storage.

closed 2023-03-08 by smagicld