smagicld/xiaoxiangBMS #223: BMS Code LIONTRON

msupermario, 2023-02-19:
Hallo, ich habe mir die Pro-Version gekauft, um meine Batterie zu schützen. Ich kann nichts einstellen. Die BMS Pin Eingabe habe ich mit allen versucht. Es springt immer wieder auf die Pin-E-Eingänge. Keine Ahnung, was da los ist. In der Demo-Version sieht die Seite der Pin-Eigenschaft völlig anders aus. Vielleicht habe ich versehentlich einen Pin falsch eingegeben? Ich komme einfach nicht auf das BMS der LIONTRON. Kann man den Pin zurücksetzen? [image] ![ 63B0ECFA-2B3D-44AE-9B09-6522D2B24463](https://user-images.githubusercontent.com/125802843/219949034-8a53004d-9c01-4b23-9536-1dee2f8d840a.png

msupermario, 2023-02-19:
Bild 1 meine Konfigurationsseite BMS

smagicld, 2023-02-19:
Ja klar kommst du nicht in die Einstellungen weil LIONTRON ein HW PIN setzt damit die Kunden keine Anpassungen selber vornehmen. Hast du die schon gefragt nach dem pin?

den PIN kann man auch mit meiner App entfernen - ist aber ein spezielles Build damit nicht jeder jedem den BMS umkonfiguroeren kann.

msupermario, 2023-02-19:
Achso! Also kann eigentlich keiner von außen an meiner Batterie rummachen? Na da brauch ich mir ja von daher keine Gedanken zu machen. Also ist die LIONTRON ja doch relativ sicher. Außer man hat vom Hersteller wohl die Pin.
Nö, die Pin habe ich noch erfragt. 
Wenn es so ist, habe trotzdem schönen Dank für deine Mühe.

msupermario, 2023-02-20:
Kannst du mir so ein spezielles Build machen, damit ich meine Batterie steuern kann wie ich will?

smagicld, 2023-02-20:
Was hat dir LIONTRON gesagt? Hast du ein Kaufnachweis?

Santana63, 2023-02-26:
Für ältere Liontron war es 619540.
Ansonsten einmal mit der rein, 
https://apps.apple.com/us/app/overkill-solar/id6444233824
danach gehts mit Jeder, Passwortschutz ist deaktiviert.
Oder schreib Dir eine App um, mit der Du das Passwort auf Werkeinstellung setzt. 
[image]

Santana63, 2023-02-26:
> Kannst du mir so ein spezielles Build machen, damit ich meine Batterie steuern kann wie ich will?

Für ältere Liontron war es 619540.
Ansonsten einmal mit der rein, 
https://apps.apple.com/us/app/overkill-solar/id6444233824
danach gehts mit Jeder, Passwortschutz ist deaktiviert.
Oder schreib Dir eine App um, mit der Du das Passwort auf Werkeinstellung setzt. 
[image]

Für JBD BMS gibt es keinen effektiven Passwortschutz mehr, das ist "Dank"Overkill endgültig vorbei.

smagicld, 2023-02-26:
Ich finde das eben sehr schlecht von Overkill diese Funktion einfach so freizuschalten - so ergibt der PIN Schutz keine Sinn. Stell dir vor jemand wird deine Batterie in Camper umkonfigurieren - das kann sehr böse enden.

msupermario, 2023-02-26:
Ich habe gerade mit Overkill die Pin von Liontron gelöscht. Hat funktioniert. Dann habe ich eine neue Pin meiner Wahl gesetzt. Somit kann ich schalten und walten wie ich will. Danke 🙏 hat funktioniert.
Natürlich kann man mit Overkill seine Batterie nicht mehr schützen. Auch wieder blöd

Santana63, 2023-02-26:
> Ich finde das eben sehr schlecht von Overkill diese Funktion einfach so freizuschalten - so ergibt der PIN Schutz keine Sinn. Stell dir vor jemand wird deine Batterie in Camper umkonfigurieren - das kann sehr böse enden.

Das schreibe ich schon seit November, da hilft nur ne neue BMS Soft mit PIN und PUK wie bei Victron Geräten. 
Leider

Das hier die Anleitung zum Umschreiben der jetzigen Hersteller APP veröffentlicht wurde war aber genau so kontraproduktiv, ich habe meine SW Versionen nicht veröffentlicht.

Santana63, 2023-02-26:
> Ich habe gerade mit Overkill die Pin von Liontron gelöscht. Hat funktioniert. **Dann habe ich eine neue Pin meiner Wahl gesetzt.** Somit kann ich schalten und walten wie ich will. 
Hast Du das auch gegen geprüft ??
Nimm jetzt mal ein Android Handy mit frischer APP, und versuche ohne Dein neues Passwort einzugeben zuzugreifen, erst dann weißt Du ob wenigsten dieser "halbe" Schutz wieder funktioniert. 
IOS APP gaukelt Dir nur nen Passwortschutz vor.

msupermario, 2023-02-26:
Hast recht 😞 öffentlich ist Mist gewesen

smagicld, 2023-02-26:
Ja… ich werde jetzt den Hinweis auch entfernen und den Button für Passwort reset Integrieren.

closed 2023-02-26 by smagicld

msupermario, 2023-02-26:
Okay

mbay0r, 2023-03-10:
> Werkeinstellung

Hat das nun geklappt, mit dem Passwort neu setzen? Danke

msupermario, 2023-03-10:
Hi, wunderbar. Alles hat bestens geklappt.Danke der Nachfrage

mbay0r, 2023-03-10:
Danke! Also BMS Pin mit Overkill löschen und dann kann ich es auch mit Overkill neu setzen, richtig?

msupermario, 2023-03-10:
Genau so 👍🏻😉

Santana63, 2023-03-10:
Seit wann hat die Overkill ne Passwort setzen Funktion?

smagicld, 2023-03-10:
nein -sie haben es einfach gemacht: es wird immer die HW Pin entfernt - schon beim verbinden an das BMS. Darum haben sie auch keine Funktion diesen zu setzen.

Finde ich echt schade, macht alle BMS unsicher.

Ich selber werde jetzt meine BMS's mit anderem Bluetooth Dongle ersetzen wo ich ein PIN auf Bluetooth Ebene setze. Bin dann sicher niemand wird mir meine Batterie umkonfigurieren.

mbay0r, 2023-03-10:
> nein -sie haben es einfach gemacht: es wird immer die HW Pin entfernt - schon beim verbinden an das BMS. Darum haben sie auch keine Funktion diesen zu setzen.
>
>
>
> Finde ich echt schade, macht alle [...]

Welchen wären das? Hast du vielleicht einen link oder eine Bezeichnung? Ich bekomme nächste Woche meine liontron Batterie und wäre daher auch an einer Lösung interessiert

smagicld, 2023-03-10:
vor ca. 3 Jahren habe ich für meine Bruder E-Bike Batterien gebaut. Bei den alten JBD's konnte man gar kein HW PIN setzen. So habe ich ein HC-05 Bluetooth Dongle genommen. den kann man über AT Commands programmieren (BT Name und access PIN). Wichtig ist den gleichen UUID zu setzen wie das original JBD BT Dongle - sonst wird der in der App nicht sichtbar.

Ich werde diese jetzt für meine Batterien umsetzen, muss aber zuerst beim Chinesen die Module bestellen. vorher will ich noch prüfen ob es etwas besseres als der HC-05 gibt.