smagicld/xiaoxiangBMS #224: App professionel

wopo22, 2023-03-03:
We bought the professinel version of the app (9,99 €), but switching the battery on and off with the doesn't work

smagicld, 2023-03-03:
This is because your pin is PW protected. Go into config section and authenticate with HW pin.
If you do not know it - I will release next week version with hw pin reset (or you drop me an email and you will get the beta)

wopo22, 2023-03-03:
The pin in the config section must be the pin of the battery?

 
 
 
 

[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage", 
"https://github.com/smagicld/xiaoxiangBMS/issues/224#issuecomment-1453490340", 
"url": 
"https://github.com/smagicld/xiaoxiangBMS/issues/224#issuecomment-1453490340", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"https://github.com" } } ]

wopo22, 2023-03-03:
> Yes. Some manufacturer (LIONTRON) set HW pin so the users won’t mess the bms settings.

The app needs pin2….What is this?

wopo22, 2023-03-03:
Attempting switching on or off we get the message „Fehler beim „Port umschalten“

wopo22, 2023-03-08:
Our battery doesn‘t work with this app. How get we back the tax of 9,99€?

smagicld, 2023-03-08:
same way as for any in-app purchase - via Apple.

closed 2023-03-08 by smagicld