smagicld/xiaoxiangBMS #225: Mistake of app

Amberwol4, 2023-03-09:
Hi everyone!

I can't turn on the discharge at Xiaoxiang BMS app. 
Every time switcher goes back. 
App purchased. 
BMS bought on aliexpress.
The seller sent only instructions.

Amberwol4, 2023-03-09:
[image]

smagicld, 2023-03-09:
Please send a screen recording. 
also the bms version + fw version details

Amberwol4, 2023-03-09:
Video will add little bit later. 
Yes I can change anything. 

[image]

[image]

Amberwol4, 2023-03-09:
[image]

Amberwol4, 2023-03-09:
While just a picture 

[image]

Amberwol4, 2023-03-09:
[Screen](https://cvws.icloud-content.com/B/Ab1qjM-Dzv39sd0fMxI5dctx-5N2AXuK6n2VZ51YvRzFnLMFqUvqBvA4/RPReplay_Final1678391887.mov?o=Ao4sLCX74Mt9L-8cTgoI6dLDe7JbSDKP5z_YED8AU7Xu&v=1&x=3&a=CAog5Xa0adS6jLkozGiAMJF2OaqCCZM36Qtc34DokAMdys0SaxDsuZHA7DAY7JbtwewwIgEAUgRx-5N2WgTqBvA4aiWzipsbrnTH8zGc5Yp3ByeJpZ0ifL9sQK_KiUoj2csE3yI5pAA-ciUzCq6qHvwXsfWoJ4ITvIxTWOdNGTAueB_px-McKy26O8qUAnJg&e=1678396574&fl=&r=9250e8b2-22ce-4d7d-a1a0-75adc06ed4cf-1&k=V7rBsc__LF2Q_-2VRgvDBA&ckc=com.apple.clouddocs&ckz=com.apple.CloudDocs&p=25&s=pV-_zMt8E4I-NXUpSwOn3pXCfKU&cd=i)

Amberwol4, 2023-03-09:
Twice was write “port success” and twice “port failed”
[image]
 

[image]

Amberwol4, 2023-03-09:
[image]

Amberwol4, 2023-03-10:
> And the B- terminal is also connected to the battery?

Yep.

Amberwol4, 2023-03-10:
All works. I'm wrong did read/save.

Amberwol4, 2023-03-10:
Thank you all very much!

closed 2023-03-19 by smagicld