smagicld/xiaoxiangBMS #226: Regression: v1.3.0 no longer shows cell voltages and BMS Infos

cnrd, 2023-03-11:
Hi

After updating to v1.3.0 the app no longer shows cell voltages and there is also information missing in the BMS Infos field
[image]
and will now ask for a password when entering config, which it did not before
[image]

I can still control the BMS relay, to turn on/off the power.

The BMSTool app still shows the cell voltages
[image]

and also correctly shows the BMS Infos
[image]

Here comes the weird part, if I open the BMSTool app and from that open the XiaoxiangBMS through the cogwheel, then the XiaoxiangBMS app seems to behave correctly
[image]

However, voltage calibration does not seem to show up, even though I did calibrate the BMS before the app update (not sure if this is expected behavior?)
[image]

smagicld, 2023-03-11:
This is very wired / was not able to reproduce it on any of our test BMS.

Have you tried to kill all the Apps and start Xiaoxiang BMS only?

Please also provide me the logs, maybe I can find something there:
1. Click on About screen -> tap the build number
2. go back, connect to bms. wait some time
3. go back to about screen -> tap the build number
4. send me the email

thanks

cnrd, 2023-03-11:
Thanks for the quick reply!

I only installed the BMSTool app when XiaoxiangBMS didn’t work, so I am sure that there was no interference originally.

I will get you the logs asap.

cnrd, 2023-03-11:
Just sent you a mail after doing the following steps:

1.: enabled debug
2.: connected to BMS
3.: waited a bit
4.: went to the config menu
5.: canceled out of the pin screen
6.: turned the battery off (which did click the relay)
7.: turned the batter on
8.: disabled debug

smagicld, 2023-03-11:
Thanks for the logs. I see in your case only INFO and Device Name is fetched, no CELL Info. I will try to find out where is the problem. will let you know once I've fond something.

smagicld, 2023-03-11:
Thanks

I’ve found the Problem. 
But are you sure it used to work before with this BMS? 
In my opinion impossible.

Anyhow I will add now support for this BMS Version, will add you as beta user so probably tomorrow you’ll get email from apple (TestFlight) with beta version to test (1.3.1)

Cheers
Lukas

cnrd, 2023-03-11:
Thanks

Completely sure, I’ve had the BMS since summer and had access to all of the features with both my current and previous iPhone.

If possible, please use the email address in the body of the email and not the one I sent the debug log from, I unfortunately was not able to sent it from my primary address (which is the one in the body).

smagicld, 2023-03-11:
ok. the 1.3.1 (build: 7662) is submitted to apple. should be released within 1 - 2 days.

cnrd, 2023-03-12:
Just got the build, working now, thanks!

[image]

closed 2023-03-12 by cnrd

smagicld, 2023-03-12:
Cool. Happy we could solve it. And if you have some time - would be great you could leave AppStore rating. Thanks!