smagicld/xiaoxiangBMS #232: Password protection on BT-Dongle

ulfertg, 2023-04-07:
On your wiki page you describe a possible swap of the BT-Dongle provided by JBD against one with password protection.
Judging from the age of the information (last edit: Nov 17, 2018) it looks as if that is something you haven't pursued any further?
I think a lot of people (including me) would be very interested in this and also willing to pay for a secure solution.

Maybe you could give us an update on this?
For example:
- do you still think this is a feasible solution?
- what would be required to continue the development?

Thank you!

smagicld, 2023-04-08:
This is the reason why I didn’t released the reset HW PIN button officially. 

I’ve ordered different BT Types to evaluate best Dongle for this BMS. I did it once 4 years ago, but can not find the manufacturer of the old Dongles. I think in 2-3 weeks I should have working solution.

For people not IT/Electronic aware i will offer service to order such BT dongle. 

Once I’m of far I will publish an update here and as release notes in new release.

ulfertg, 2023-04-08:
That's great news!
In case you need a Beta-Tester: I'd be happy to help!
I'm currently running two BMS systems, a SP04S034 (4S 200A) and a SP04S028 (4S 150A).
Can't wait to hear more about this!

While this sounds like the perfect solution for the community of DIY battery builders, I'm really curious how the big companies like Liontron will tackle the current security issue with JBD BMS.
I'd bet they'd try anything to avoid recalling all their batteries out there, but maybe they'd be interested in retrofitting a secure BT-Dongle in the batteries they are going to sell from now on?

smagicld, 2023-04-29:
Hi. Just FYI. Today I’ve received the programmable BT Dongles. 

tomorrow I’ll start coding dedicated app for it. 
the idea is that anyone can order the dongles on AliExpress and with the App you can provision the Dongle (set BT Pin, BT Name) and make JDB compatible.

will keep you updated. 

[image]

ulfertg, 2023-07-03:
Hello,

how is it going with this project?
I'd love to hear an update if you have some time to share one :)
Thank you!

smagicld, 2023-07-05:
Hi

Yes! very good news.
After doing tests with lot of BT Dongles finally I've found one working and easy to setup!
The name of BMS is then set directly on the dongle as well 6 digits long PIN code. The authentication is done on iOS/BT connection level - not the App. so it is real HW protection.

If you drop me an email I can provide you details which Dongle to buy and how to setup it.
(You will need an USB <-> RS232 that supports 3.3V).
Next week I'll try to prepare an web page with more details.

[image]
[image]

[image]

ulfertg, 2023-07-05:
That's great news!
I'd love to hear about the details and the specific dongle - my mail address: porches.headset.0f@icloud.com (a temporary one, since I didn't find a way to send it other than posting it publicly).
Thanks in advance!

bowmore16, 2023-10-15:
Hi, can you provide some information on the preferred BT-Dongle and the software?
I would like to order the dongles to proceed with the project.
Many thanks, Martin

f-io, 2024-10-31:
Are there any news regarding a password protection?
I'm using the far right BT dongle together with a SP15S008C-15S (FW19) -> App: Firmware too old for HW Pin.