smagicld/xiaoxiangBMS #233: Config data incomplete with older BMS

ulihuber, 2023-06-02:
I have two virtually identical sytems. The newer still works fine on the iPhone app but the older now shows incomplete configuration data.
[image]
Was there any  update with compatibility issues in recent time?

smagicld, 2023-06-02:
Hi

Es liegt an der Firmware. Bei älteren Firmware ist es nicht möglich so granular die Spannung / SOC Konfigurieren. 

Die App versucht immer bei jedem Firmware alle Felder auszulesen. Sollte der BMS bei einem Feld als „Unbekannt“ antworten - so wird das EditBox nicht angezeigt. 

Firmware Version siehst du auf dem BMS Übersichts Screen.

smagicld, 2023-06-02:
welche Firmware Version hat der BMS der nicht alle % Felder anzeigt? <20

ulihuber, 2023-06-02:
Firmware ist 17
Die hat aber immer mit der App funktioniert