smagicld/xiaoxiangBMS #238: turn off, turn on charge/discharge doesn’t work

Kass271, 2023-09-11:
Looks like when I click toggle switch, it doesn’t store on UI part it status. I added my screen video.
This relates for bms on volt Polska liFePo4 battery

smagicld, 2023-09-11:
Port change requires open BMS or authenticated if Hw pin is set. 

Looks like your BMS has HW Pin set. Can you go into Config and read them out?

After config readout / PIN entry go back to main screen and toggle the port.

Kass271, 2023-09-11:
Thank you for fast reply!!
I added photos

Kass271, 2023-09-24:
@smagicld so after I tried different steps, problem still the same. Looks like togle reset status before it will send to bms

smagicld, 2023-09-24:
You can send me the debug logs. I will take a look on it - but in two weeks. Going for holidays.

ysmolski, 2023-11-11:
Same issue here with 100A LiFePO from Volt Polska. Cannot toggle the switch for charging/discharging ports. It goes back. Please let me know how can I collect debugging logs for it? 

[image]
[image]
[image]

smagicld, 2023-11-11:
Hi @ysmolski 

1. Go to about screen and tap on the version / build number once:

2. it should be confirmed:

3. go to main screen. try to connect to your BMS. Try to switch ports on/off. 

4. go back to about screen and tap on version / build number

5. send me the logs over email

Dillon1630, 2023-11-11:
I’m having similar issue, toggle will not stay set and I can’t clear alarms

smagicld, 2023-11-11:
> I’m having similar issue, toggle will not stay set and I can’t clear alarms

some battery manufacturers just set HW Pin. if this is set, you can not toggle ports on/off. in this case go into config. and if you'll be asked for HW PIN - enter the correct HW pin (ask your battery seller what they set for your battery)