smagicld/xiaoxiangBMS #239: Waiting data on Ant BMS

uffo95, 2023-09-22:
[image]

it keeps there forever. any solution or am I missing something. thanks

Franceskoooo, 2023-09-24:
Hi exactly same problem... I just now bought PRO feature and nothing changed...So I hope you will fix it... I tried on iOS 15.7 and iOS 17 - no difference
[image]

Franceskoooo, 2023-09-24:
@smagicld  please see the comment up, thank you, your all is the best of all! 
hope it will work soon.
Best regards

smagicld, 2023-09-25:
The Xiaoxiang BMS supports only xiaoxiang BMS. No Ant BMS or so. It’s totally different bms / manufacturer. 

You can try the BMS Tool app to read Ant BMS devices.

Franceskoooo, 2023-09-26:
I see, thank you I didn’t realised that... would you add support for this Ant BMS in future maybe please? I love BMS Tool app so please take that PRO feature i bought as compensation for awesome app! :) Would be awesome to have more functions in BMS Tool app. 
Thanks man!

closed 2024-06-27 by uffo95