smagicld/xiaoxiangBMS #240: BMS software not properly working on iOS 17 on new iPhone 15 Pro.

GTVR129, 2023-10-02:
Keeps searching for and creating false devices. At same time it looses my payment history and keeps telling me to buy the app. I restore the purchase and it works for a day and starts issue again. Cannot have this. I must be able to monitor battery status very regularly. I have already deleted and reinstalled the app. Same issue. Please advise.

smagicld, 2023-10-02:
Hi. This bug popup sometimes. Delete the app, install again from AppStore and restore purchases. This should help.

GTVR129, 2023-10-02:
Already deleted the app and reinstalled it. Restored purchase. Same issue.

GTVR129, 2023-10-02:
Reinstalled a third time. Let’s see what happens. Thanks.

smagicld, 2023-10-02:
Ok. Just tried with the latest 17.0.2 and can reproduce the issue. Will provide asap an fix!

GTVR129, 2023-10-02:
Ok thanks let me know.

smagicld, 2023-10-03:
Good morning. Just FYI: The issue is related with iOS 17 only and we believe it is n apple bug. 
anyhow, we've made an workaround for this issue. Version 1.3.2 was just submitted into AppStore. should be released today / tomorrow. 
l

GTVR129, 2023-10-03:
Thanks for your quick solution. Installed it so far so good. I’ll let you
know if anything.