smagicld/xiaoxiangBMS #243: Port Change Failed and Reading data from bms

denis5jlion, 2023-11-23:
So i cant change ports any more on my bms, and reading data from bms is like 1 in 15 time. Pls fix, it has something to do with password or pro version?

smagicld, 2023-11-23:
It looks like BT range / signal is weak. 

Which previous version you mean? Please provide version number.

denis5jlion, 2023-11-24:
I tried with phone 5cm to bms its same. What about port change failed ?

smagicld, 2023-11-24:
Since more than one year there was no code change in the BT connection part. so it can't be the App version.

What is the issue with port change?
please provide Logs, Error messages, Screenshots, iOS and App Versions so I can try to reproduce your issue.

desultorynz, 2023-11-27:
@smagicld yes I too have this same issue.  The port change works fine UNTIL I add a hardware pin onto the BMS, and once I do that and then close the app and re-log back into the app and enter in the correct pin then the PORT Change function will NEVER work correctly again.  Eg I can't turn off the charge or discharge ports at all once there's a hardware pin in place even when I'm using the pin, yet until I add the hardware pin it works fine.   Please let me know what you need from  me thanks, as clearly its broken and needs fixing. I'm Ios 17.1.1 latest ios app version v.1.3.2  BMS model dP04s007L4S150A.

desultorynz, 2023-11-27:
@smagicld [image]

smagicld, 2023-11-27:
The Issue is that the PIN is set. you should go into config and enter valid PIN when asked. 
Then the port on/off will be possible.

closed 2023-11-27 by smagicld

desultorynz, 2023-11-27:
@smagicld No, I had set the pin myself and then re-connected to the BMS and entered the correct Pin, it allows me to change the BMS release values and write updated values to the BMS (which it wouldn't allow if the pin wasn't correct) yet it still does NOT allow me to turn off the MOSFETS, even with the correct pin, hence as I said its a software / pin remembering issue still not resolved, not just the wrong pin as you are claiming.

desultorynz, 2023-11-27:
> The Issue is that the PIN is set. you should go into config and enter valid PIN when asked. 
> 
> Then the port on/off will be possible.
> 
> 

Have done just as you said and entered correct oin and even removed the pin and re set it correctly again and it still then would never let me turn off the mosfet. Tried on 6 different batteries all with same BMS model and it didn't work once the pin was enabled even with the Correct pin entered.

desultorynz, 2023-11-27:
> So i cant change ports any more on my bms, and reading data from bms is like 1 in 15 time. Pls fix, it has something to do with password or pro version?
> 
> 
> 
> PS. previous version worked fine

Yes the same issue for me too once a pin is used then even the data reading is not so reliable even though iv re entered the same pin that I had just set. Something is strange just only affecting things working correctly only once the hardware pin function is used. As soon as I clear the pin it all works perfect until I re add a new Hardware pin again, it always used to work fine I'm not sure which version broke it sadly.

smagicld, 2023-11-27:
I have 15 different JBD BMS including latest one (with production date from June this year). I’m not able to reproduce your issues on any of them. 
therefore please provide me step by step guide how to reproduce your issue. 
please also provide me your BMS model number, app version and iOS version you’re using.

desultorynz, 2023-11-28:
@smagicld If you need any further info from me about this software bug / issue please let me know what process to run or use to help you further thanks.

denis5jlion, 2023-11-28:
Hi i sent log to xiaoiang@e77code.com. I am using Iphone 12 pro with lastest version of ios. Changed pincode. Using qucc bms.

desultorynz, 2023-11-28:
> Hi i sent log to [xiaoiang@e77code.com](mailto:xiaoiang@e77code.com). I am using Iphone 12 pro with lastest version of ios. Changed pincode. Using qucc bms.

Did you get it sorted and working with the latest version of the app when using a hardware pin code, or still having issues?  Once I used the pin removal tool then all functions worked fine, but as soon as I re-added a new pin and then logged out and re-opened the app and re-connected and then entered in the pin when asked then it still wouldn't allow the mosfet to be turned off, even though the correct pin was entered, the only way I could make it work again was to use the pin removal tool and fully remove the pin.  It seems some kind of bug only affecting usage with a pin.

smagicld, 2023-11-28:
Hi @desultorynz 

1. No email with logs received on xiaoiang@e77code.com. please re-send.
2. no time yet to investigate your issue. I'll let you know when we have some news.

denis5jlion, 2023-11-28:
Hi i resend it now with different addres, check if you got it @smagicld

denis5jlion, 2023-11-28:
@desultorynz Where do i get pin removal tool so i can remove pin at least so it works?

smagicld, 2023-11-28:
Hi @desultorynz 

1. the email you are sending is wrong. one x is missing.
2. I tried to reproduce your steps but it works for me. BMS from 26.08.2023 with FW Version 61.

But one thing in your description is suspected: "... and Set Pin (hardware pin) and set pass code as 345678 and press okay. Now go back to device list and connect to the bms name again and then click config and enter in the same pin 345678 when asked for it, ..."

if you set the HW Pin, then it is stored locally in the phone. if you re-connect and go into config - then the stored PIN is taken - so you do not have to enter it every time. This is strange. 

Please try following: delete the app from the phone (with all data) and install again from App Store. set the HW Pin, disconnect, connect and go to config. is the PIN Dialog still popping up?

desultorynz, 2023-11-28:
> Hi @desultorynz
> 
> 1. No email with logs received on [xiaoiang@e77code.com](mailto:xiaoiang@e77code.com). please re-send.
> 2. no time yet to investigate your issue. I'll let you know when we have some news.

I've just re-sent you the debug.zip file again  hope you receive it this time thanks.

desultorynz, 2023-11-29:
> Hi @desultorynz
>
> 1. the email you are sending is wrong. one x is missing.
> 2. I tried to reproduce your steps but it works for me. BMS from 26.08.2023 with FW Version 61.
>
> But one thing in your [...]

I don't want to delete the App with ALL data as I have literally more than 30 saved configs in the data, and if I delete all the data then I'd lose all of those (which I don't want to do), unless you can offer me some way of backing those up and then restoring them again after reinstalling the app?  I've also tried on two different iphone devices and both have the same issue but are both running the same ios app version, even though one phone is ios 16.2 and the other is 17.1.1

desultorynz, 2023-11-29:
@smagicld is there any way at all that I can SAVE or BACKUP the saved bms configs from my iphone, so that I won't lose them if I delete the app and then re-install it, so that I can then restore the "config backups" back to my phone, as I use those weekly hence they're critical I have access to them.  I can't find any way of backing this up using icloud or even if I go into file manager on the phone I can't see anywhere for your app where I can save config files etc, so perhaps this is something (some way) you can add into a future version to have the ability to backup config files to icloud or elsewhere, just for times where a full app delete and re-install is needed, yet the config files need retaining, thanks.

smagicld, 2023-11-29:
@desultorynz all files can be backed up. Just connect the phone to the Mac, go with finder into the phone -> app -> documents and you can drag and drop the files. 

I really have no idea how to help you with the other issue because it is impossible for me to reproduce it. 

Last chance is the debug file - but I still did not received it.

desultorynz, 2023-11-30:
> @desultorynz all files can be backed up. Just connect the phone to the Mac, go with finder into the phone -> app -> documents and you can drag and drop the files.
>
>
>
> I really have no idea how to [...]

Is it possible to backup the files without a MacBook eg using a pc with windows ? Or even directly from the iPhone to a cloud service such as google drive or any others?

Iv emailed the debug file twice, not sure why your not getting it, is there another place I can upload it to such as ftp or a drive cloud service etc?

smagicld, 2023-11-30:
Finally Ive got the debug file. will answer you via email. 
if you enter in google: " how to access iPhone app documents on windows" you'll tons of guides.