smagicld/xiaoxiangBMS #244: SOC drift

ken-am, 2023-11-25:
My soc keeps drifting off to where it is indicating a higher soc than what is actually present. If I reload the config, it comes back in line with the expected soc but then drifts off after only a few charge/discharge cycles. Is this an error in my config?

sarah1331sarah, 2023-11-26:
When you load a config it’s setting SOC from battery voltage (not the best way) then it uses the columb meter to measure (the more accurate) best way it to load your settings and then fully charge the battery

ken-am, 2023-11-26:
I understand that but after a few charge/discharge cycles it's indicating that the SOC is higher when the battery has been exhausted.  Is there something in the config that I should look at?  How does the columb meter know that it takes more to charge due to losses from heat (charge efficiency) versus discharge?
Ken

sarah1331sarah, 2023-11-26:
Assuming all your settings are correct your battery is worn but I would suspect settings first

ken-am, 2023-11-28:
Assuming the settings are the problem, do you have any recommendations where to start looking?
Ken

sarah1331sarah, 2023-11-28:
Post a copy of them

ken-am, 2023-11-30:
Is there a way to export them or do I have to take screenshots?Sent from my iPhone

sarah1331sarah, 2023-11-30:
Screen shot it but you will have to do it on GitHub not replying to messages

ken-am, 2023-12-01:
I'm not sure how to do that.
Ken

sarah1331sarah, 2023-12-01:
Follow the link in the email

ken-am, 2023-12-02:
I'm not sure if this is how you meant but I didn't see any link from your iPhone.

[image]
[image]
[image]
[image]

sarah1331sarah, 2023-12-02:
What battery chemistry do you have

IMG_0385.PNG (view on web)<https://github.com/smagicld/xiaoxiangBMS/assets/152033122/6e2bd561-0c88-4136-9d0d-f595c37928e7>
IMG_0386.PNG (view on web)<https://github.com/smagicld/xiaoxiangBMS/assets/152033122/540dde00-56e7-4eb4-9a57-541053556d25>
IMG_0387.PNG (view on web)<https://github.com/smagicld/xiaoxiangBMS/assets/152033122/fdb38cc0-509f-4c19-8d53-c7ed31eda54e>
IMG_0388.PNG (view on web)<https://github.com/smagicld/xiaoxiangBMS/assets/152033122/61fd0ae3-4da6-43fb-9c2a-ce419fabbb83>

sarah1331sarah, 2023-12-03:
Your total wants to be 96000
Your cycle wants to be 75000

if the battery is used i would minus 10-15% off these too

then fully charge not so the bms shows full so your charger hits float

ken-am, 2023-12-04:
Ok! I originally had the total at 96000 but increased it to see if that would solve the drift problem.  I'll try these settings for a while and let you know how well they work.  Since I am using these in a solar system, I haven't been able to reach float mode for a while since it's been overcast a lot lately.  Today for instance they only got to 62% from 21% last night, well short of the float voltage requirement.  Could you give me some insight into the cycle capacity?  It's a bit of a mystery.
Ken

sarah1331sarah, 2023-12-04:
Cycle capacity is the “useable” capacity of the battery lead acid is 50% for example lithium is 80% 

but if the batteries are worn then the useable capacity will be 60%but this could be found by load testing them 

solar is rubbish I have 400ah battery and only getting 0.01-0.02kwh a day from 330w solar

ken-am, 2023-12-09:
I agree.  I have 4 245w panels in a 2S2P configuration out on my deck, not on my roof, and I am finding it difficult to fully recharge my 192ahs every day after using them the night before.  Last night for example, my batteries were at 21% and today as I return home from work, they are only at 85%.  However it was a rather gloomy day with little clear skies so there's that.  330w on a 400ah battery seems a bit undersized to me but if you don't have much of a load and are looking for long time emergency power with a low and slow recharge, then I guess you're ok.
Ken

sarah1331sarah, 2023-12-09:
Caravan roof space is limited ;)

ken-am, 2023-12-10:
That explains it.  I have another question.  What happens when the SOC reaches 100% before the cell voltage reaches the max limit?  Does the columb counter keep adding to it's total or does it stop essentially resetting the SOC max?  Does the BMS stop charging or continue until the max cell voltage is achieved?
Ken

sarah1331sarah, 2023-12-10:
i cannot comment for all BMS's but most reach 100% and reset its "cycle capicity" regardless of genuine status. obviously it will cut if OV or COV but this is why after a program the BMS uses the values entered to "guess" the SOC its wise to fully charge and hit float with your battery if you edit the settings again you should do this to make an accurate measurement 

i had a victron shunt and my bms over the 6month period they kept within 0.5% of each other so i did away witht the shunt and connected the BMS to my cerbo 

as prior i didnt beleive the bms to be as accurate as most dont register <5watts and on standby my caravn used 4~7w

ken-am, 2023-12-12:
Ok, so if I understand you correctly, the SOC will automatically reset once it reaches 100%.  That would explain why it has drifted so much as of late, since the weather has not allowed my batteries to reach 100% for several weeks.  Right now as I came home, I found them at 99% and only 11.50 volts while a full charge should be in the neighborhood of 12.6 volts.  Maybe tomorrow they will finally get to 100% SOC (weather permitting) and I will be able to confirm the reset of the SOC.
Ken

sarah1331sarah, 2023-12-12:
Have you a mains charger to charge and get to float to reset

ken-am, 2023-12-14:
Yes but it isn't connected as I want to use total solar to maintain the batteries.
Ken