smagicld/xiaoxiangBMS #245: Waiting data after clear the PIN protection

sirszabo, 2023-12-21:
Hello,

I have two pcs JBD-AP21S001-L21S-200A-B (v1.2) BMS, where the pro application worked without any issue till I have cleared the PIN protection.
After I just showing the "waiting data" when I try to connect to the BMS. The "blue elephant" XiaoXiangEletric application stopped working on both devices also! 
So, I have connect by PC and I saw the "Model" is missing on the configuration section. 
When I writing back to "JBD-AP21S001-L21S-200A-B" (I hope this was the default, because unfortunately I didn't note it?!) the "blue elephant" XiaoXiangEletric application started to working, but the application can read the model name as "JBD-AP21S001-L21S-200A-**ú**" ... and the connection still not working with XiaoxiangBMS ... I can see "waiting data".
Then I have modified again the model name, because the "-ú" character was not okay for XiaoXiangEletric application, I can't manage the BMS. 
Right now I have added the "JBD-AP21S001-L21S-200A" as model name and the XiaoXiangEletric application started to work normally, but the XiaoxiangBMS unfortunately not.
Could you please help for me, what is happened after I cleared the PIN protection and how I can use again XiaoxiangBMS application?

Many thanks!

sirszabo, 2023-12-24:
If I set back to "NONAME" doesn't help! Please help me to solve this issue, because this is a very frustrated problem, where "Xiaoxiang BMS" application did some modification on my BMS's and I can't restore it! I can access to the BMS's via PC, but this is not a solution!

sirszabo, 2023-12-24:
I have attached the pictures from the BMS applications ...

[image]
[image]

sirszabo, 2024-01-07:
Okay, so pay for nothing ... and of course, no support .... aaaand a shi*****ty application did a serious modification on the BMS WHERE the manufacturers can't help (yeah, I have talked with the JBD manufacturer about this)!! Brilliant ... DO NOT PAY FOR THIS APPLICATION!!!!!!!!!!!!!!!!!!

smagicld, 2024-01-07:
Hi. please send me the details of your conversation with JBD to xiaoxiang@e77code.com.
Please also provide step by step instructions how to reproduce your issue and also describe the serious modifications.
Please also provide the App Version you're using.

we've tested setting and removing PIN and it works without any issues. setting name to JBD-AP21S001-L21S-200A-B worked too (screenshot attached). 

[image]

desultorynz, 2024-02-03:
> Hi. please send me the details of your conversation with JBD to xiaoxiang@e77code.com.
>
> Please also provide step by step instructions how to reproduce your issue and also describe the serious [...]

Are you able to message me back on WhatsApp and you can let me know if you may be able to further assist me, and what price it would be for your assistance thanks.

smagicld, 2024-02-03:
sorry, I do not provide any personal assistance.

closed 2024-02-03 by smagicld