smagicld/xiaoxiangBMS #247: Charging cycles are not displayed...

Peliccan, 2023-12-28:
...without reading the BMS settings first. The cycle display then works until the app is closed.

Peliccan, 2023-12-28:
App -> 1.3.2
iOS 17.2.1 -> iPhone 15 Pro

smagicld, 2023-12-28:
can not reproduce. cycle count is read / shown after connecting to bms. 

please attach / send via email screen recording.

Peliccan, 2023-12-28:
Anhang verfügbar bis 27.01.2024
Zum Laden klicken
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAVHtelomePvOCZYbmn9uPZAt8DP9AcAygIq_DhJTb6SJuwmHtOB8qgEL%2F%24%7Bf%7D%3Fo%3DAhBti1vfNUPtR62h7fXQBHqe0_GyZC67Qk4CrcoD1v9T%26v%3D1%26x%3D3%26a%3DCAognBuZqHBIvPuYFRDasl0LLPrYApedCKavnxYuybg2ntcSeBD6ovGOyzEY-rLs4tQxIgEAKgkC6AMA_wzVvUlSBC3wM_1aBHyqAQtqJqCOCu_Rd3yNNlwgyDGGUTcsm7yS-FvYFXpj-7OeAm7ViBug3F4LcibS1iJLv5kuOveD7-z-MRqNj5uAs0gD3Z5mHzIU7m1xfinKR3eUGQ%26e%3D1706383055%26fl%3D%26r%3DB204F6E8-EE63-4F55-B8E9-F8A0B2BD50E6-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3D89523B53-93A5-4253-8E04-BF8C3C14227C%26p%3D41%26s%3DJ3pPyJt-anNq5RPeqyE2-HK5rdE&uk=AyTZIB5wymRUTfGETC7Spw&f=Demo.mov&sz=79851264>Demo.mov
79,9 MB

smagicld, 2023-12-28:
danke für die Bildschirmaufnahme. Das verhalten ist typisch wenn das Signal zu schwach oder gestört wird. Du kannst versuchen den BT Dongle anders zu platzieren oder mit einem anderen iOS Gerät sich verbinden. 
Ich kann das verhalten reproduzieren nur wenn der BMS/BT Dongle sehr weit vom Mobile entfernt ist.

closed 2023-12-28 by smagicld

Peliccan, 2023-12-28:
Ok. Aber das Telefon liegt genau auf dem BMS. Näher dran geht nicht. :)

Peliccan, 2023-12-28:
Habe das auch mit einen anderen iPhone getestet. Das Verhalten ist immer gleich.

Peliccan, 2023-12-28:
Aber gut. Vielen Dank trotzdem für die schnelle Reaktion + die sonst sehr gute App!!