smagicld/xiaoxiangBMS #263: App not responding/refreshing

DonnieL60, 2024-11-27:
App is not responding/refreshing with with actual battery charge level. Not sure how to check what version I’m running.

smagicld, 2024-11-28:
and what is your question?
if you have issues, please provide steps how to reproduce, your BMS HW Type, ios version and some screenshots.

vrooomie, 2025-02-14:
I have a VATRER 460ah battery, and it will not respond to the Xiaoxiang BMS. I have a Temgot 470, and the app works fine with it.

 I have done everything that I possibly can, including talking to Xiaoxiang and Vatrer and they disavow any ability to help me.

I am using an i15 Apple phone.

smagicld, 2025-02-14:
Hi 

What is VATRER 460ah? Please provide model of the built in BMS (need the model name and exact type. Also firmware version. On all XiaxiangBMS you’ll find this info on the label).